NWFSC-OA-lab/asw #1: Plumbing

pmcelhany, 2021-05-27:
- [ ] DI to HS: complete pipes from wall to HS tank (elbows, ball valve, float valve)
- [ ] HS to Head tank: complete pipes from HS to wall (from bottom 1/5th of tank)
- [ ] HS to Head tank: HS pump (Mike order, adapt plumbing for connection)
- [ ] HS to Head tank: final pipe connections at head tank
- [ ] Install DI filters
- [ ] DI to head tank: plumb in solenoid (wire solenoid) and final connections to head

dmperez89, 2021-06-02:
@pmcelhany @michaelmaher114 
Something to think about for our meeting today: Are we leaving the lid to the rain barrel/HST on (with holes drilled in it for the pipes?)? Or will it be left open?

pmcelhany, 2021-06-02:
@dmperez89 @michaelmaher114  
A lid seems like a good idea so stuff doesn't fall in the tank and to reduce evaporation. We need to be able to take off the lid, so maybe the pipes into the tank are attached with unions

pmcelhany, 2022-03-28:
Combined with another issue

closed 2022-03-28 by pmcelhany