NWFSC-OA-lab/asw #2: Sensors and Controls

pmcelhany, 2021-05-28:
- [ ] conductivity sensor with comm card (HS tank) **need to order**
- [ ] temperature sensor (HS tank) - evaluate the conductivity temperature probe
- [ ] conductivity sensor (head tank) install & order communication card
- [ ] durafet and UDA - install
- [ ] temperature sensor (head tank) - evaluate the conductivity and durafet (are durafets fast enough)
- [ ] Heaters (HS and head tank)
- [ ] labjack (only if needed for temperature)
- [ ] Ethernet switch (check with Tony or **order**)
- [ ] Relays in enclosure (for 2 heaters, HS pump, DI water solenoid) have the relays (**find enclosure**)
- [ ] 24 VAC transformer in same enclosure as relay for HS pump - **find or order**
- [ ] computer connection (pick computer and monitor)
- [ ] water level sensor install USB

pmcelhany, 2021-11-01:
@michaelmaher114 Mike, lets review where we are with the artificial seawater system. I think we need to update this issue...

michaelmaher114, 2021-11-01:
Sounds good.  I think I'm going to try and go out to Manchester on Wednesday probably but I am available the rest of the week.

closed 2022-03-28 by pmcelhany