NWFSC-OA-lab/asw #5: SOP for salinity probe

pmcelhany, 2021-06-21:
Should we write up a short OA lab SOP/guild for the salinity probe describing calibration, settings, troubleshooting, etc.?

dmperez89, 2021-07-02:
My SOP now lives [here](https://docs.google.com/document/d/14IsDH_83C37d63tQa3EZf1kIPZ7zOBKpgjqCsFmWSig/edit): 
Ocean Acidification Drive --> 03. OA Systems, Experiment Chemistry & Equipment--> 01.  SOPs --> ThermoScientific Orion Star A322 calibration SOP

pmcelhany, 2021-07-08:
Hi Danielle - 
Thanks for drafting this. I made a few comments. At some point, we will want someone who has not calibrated the sensor before (Kathleen?) to give the SOP a try to see if there is any missing info.

pmcelhany, 2022-09-26:
going to reopen until this is finalized

pmcelhany, 2022-10-24:
@dmperez89 add the step of measuring the conductivity of standard before calibration. This will give us a record of how much the instrument has drifted since the last calibration. We also need to find a better place on the OA drive for the salinity calibration log...

dmperez89, 2022-10-31:
I added the step of measuring the standard before calibration to the SOP. I also created a new tab in the calibration log for ongoing probe measurements to look at drift. I moved the log to be in the same [folder](https://drive.google.com/drive/u/0/folders/1UaG3ILah7ACxrgPCQpb9dHjGlz4Panq2) as the SOP, but I'm still not sure if this is the best place....

pmcelhany, 2022-10-31:
@dmperez89 How about we just put it [here](https://drive.google.com/drive/folders/14SURnhKceJB9coT7WP7DxHl8MI9YlnCK?usp=share_link) in a new folder.  There are really too many subfolder (16) so we will need to add another level of hierarchy to this at some point, but a folder named salinity seems pretty transparent for now.

dmperez89, 2022-11-01:
Done, the calibration log is now in the salinity folder within systems, experiment chemistry, and equipment. I left the SOP in the SOPs folder for now.

pmcelhany, 2023-08-21:
completed

closed 2023-08-21 by pmcelhany