NWFSC-OA-lab/asw #8: CO2 alarm

pmcelhany, 2021-09-03:
We need to order a CO2 alarm for the lab. 
[Here](https://www.huffpost.com/entry/gassing-satartia-mississippi-co2-pipeline_n_60ddea9fe4b0ddef8b0ddc8f#:~:text=On%20February%2022%2C%202020%2C%20at,CO2%20gas%20near%20Satartia%2C%20Mississippi.&text=Some%20experts%20estimate%20this%20network,miles%20of%20existing%20petroleum%20pipelines.) is some reporting on why CO2 is bad.

dmperez89, 2021-09-14:
Some options--all have audible alarms, but none say anything about being approved by FDA:
[Extech (available on GSA)](http://www.extech.com/products/CO220)
[Grainger](https://www.grainger.com/product/21RV54?ef_id=CjwKCAjw7fuJBhBdEiwA2lLMYaR5oSnKBuEAWZoHa6YJLteO6c88rDXR7yKuajZoyReTfpgiBEdIxBoCEycQAvD_BwE:G:s&s_kwcid=AL!2966!3!496359978193!!!g!438000512349!&gucid=N:N:PS:Paid:GGL:CSM-2295:4P7A1P:20501231&gclid=CjwKCAjw7fuJBhBdEiwA2lLMYaR5oSnKBuEAWZoHa6YJLteO6c88rDXR7yKuajZoyReTfpgiBEdIxBoCEycQAvD_BwE&gclsrc=aw.ds)
[Gain Express/Amazon](https://www.amazon.com/Temperature-Humidity-Mountable-Detector-0-2000ppm/dp/B00MB93CK8/ref=sr_1_13?dchild=1&keywords=Co2%2BAlarm&qid=1631575963&sr=8-13&th=1)
[Forensics Detectors/Amazon](https://www.amazon.com/Dioxide-Monitor-Forensics-Compliant-Classrooms/dp/B08XH5ZG71/ref=sr_1_5?dchild=1&keywords=carbon+dioxide+alarm&qid=1631576023&sr=8-5)
[GZAIR/Amazon](https://www.amazon.com/GZAIR-Temperature-Humidity-Tabletop-Mountable/dp/B0986Y7PVS/ref=sr_1_8?dchild=1&keywords=carbon+dioxide+alarm&qid=1631576023&sr=8-8#customerReviews)
[Langkou Store/Amazon](https://www.amazon.com/Detector-Pollution-Temperature-Humidity-Professional/dp/B08Q7YMX9X/ref=sr_1_14?dchild=1&keywords=carbon+dioxide+alarm&qid=1631576023&sr=8-14)

pmcelhany, 2021-09-14:
Thanks for tracking these down. Maybe certification is from OSHA. What do
you recommend?
-paul

dmperez89, 2021-09-14:
I didn't see anything about OSHA, or really any approval. The one from GSA might be good if it makes things easy for Kathleen. Do we care about wall vs battery powered, or connecting to Wifi? Or do we really just want something that's going to beep really loud if we go above a certain CO2 level?

pmcelhany, 2021-09-14:
mainly beeping. Which one is gsa?

dmperez89, 2021-09-14:
[Extech on GSA](https://www.gsaadvantage.gov/advantage/ws/catalog/product_detail?gsin=11000038532310)
Apparently it's on sale! But only until September 30

pmcelhany, 2021-09-14:
I like that it has a data logger. Too bad we will miss the sale!
-paul

dmperez89, 2021-09-14:
Sounds good, I'll put in a request to Kathleen

pmcelhany, 2021-09-14:
Can't make any purchases until we can spend money again in October. It is
good to have found an option that will work so we are read to go when the
funds arrive.

dmperez89, 2021-09-15:
[Google Doc purchase request for Kathleen ](https://docs.google.com/document/d/1lFNZi-9C7qWob79vV0HXBioEF_2npXI84xHiOrE-yyw/edit?usp=sharing)

Ocean Acidification Drive -> 06. OA Lab Logistics -> 01. Purchase Orders, Receipts, & Packing Slips -> Purchase Orders

pmcelhany, 2022-03-28:
installed

closed 2022-03-28 by pmcelhany