NWFSC-OA-lab/asw #22: DI water system

pmcelhany, 2022-04-21:
- [ ] Evaluate current status of the DI water system (e.g. RO system flow rates)
- [ ] Decide next steps for the system

pmcelhany, 2022-06-02:
@KROVINSKI @dmperez89 @michaelmaher114 
I replaced the filters and installed the new pressure booster pump. The both of these things make a big difference - we now get pretty decent flows from the RO system. I have left the booster pump unplugged when I left because I haven't installed the pressure switch yet and didn't want to risk burning out the pump. However, it is good to keep water flowing through the system every once and a while. So, can you plug in the pump and push a little water through the filters for a couple of each minute each day you are in the lab? Just connect the pump to the connector coming from what looks like a laptop power supply in the window. The water should fill the pressure tank so no need to mess with any valves. Here are some other next steps

- [ ] periodically run the booster pump to move water through the system.
- [ ] Confirm the contents of the D.I. filter housings. Are there filters in there? When were they installed?
- [ ] Connect the second pressure tank. Looks like we just need a tee push-connect fitting for the first pressure tank. Do we have one or need to order from mcmaster?
- [ ] test and install booster pump pressure switch (the switch is currently in my laptop bag at home) I can do this next time I'm at the lab.
- [ ] Once we have water flowing through all the filter, collect a sample in metals-free bottle (like for crab isotopes) and send to alex for analysis.
- [ ] Once we have water flowing through all the filters, measure conductivity with salinity meter and compare to tap water. Make sure filter water is in clean container and salinity probe is rinsed very well.

pmcelhany, 2022-06-03:
@KROVINSKI What is the current status of the DI filters?

KROVINSKI, 2022-06-03:
I labeled the filters and didn't get to maintenance early enough to ask to borrow a strap wrench to take off the DI housings. The housings have the O-rings loose so I don't think they are in there. I image the DI filters are the ones bought in April 2021.

pmcelhany, 2022-06-03:
@KROVINSKI OK. Let's install them next chance we get. I ordered DI filters from US Water Systems on 2021.05.04 - I assume those are the filter sitting on the counter?
Also, do we have the push-connect tee so we can add the second pressure tank? If not, let's order from mcmaster.

KROVINSKI, 2022-06-08:
@pmcelhany connected the second pressure tank (2022.06.05). 
Switch test- unk
CONDT- tap 61.23 micro siemens / cm, 30.54 PSU @ 18.8C
CONDT- filtered water 54.66 micro siemens / cm, 27.33 PSU @ 18.8C

pmcelhany, 2022-06-09:
@KROVINSKI @dmperez89 
Table 1 on [this site](https://www.teachengineering.org/activities/view/uoh_drinkthat_activity1) shows expected conductivity of different types of water. To compare tap water, our filter system water and Milli-Q we need to range from ~800 uS/cm to <0.1 uS/cm. Seawater is ~55,000 uS/cm so those standards are  in a different range. We could use standards at [25 uS/cm](https://www.fishersci.com/shop/products/conductivity-standard-25-mhos-cm-milliporesigma-supelco/111004897) and [1,412 uS/cm](https://www.fishersci.com/shop/products/conductivity-standard-1-412-mhos-cm-milliporesigma-supelco-2/111016314) to calibrate. (25uS is the lowest standard I found in my very quick search -lower would be better). Conductivity should give us a pretty good indication of how well our filter system is doing. I will also see if we can get additional purity info from Alex's mass spec. Please ask Kathleen to track down the standards.

KROVINSKI, 2022-06-10:
@pmcelhany, okay will send Kathleen an email, do the GSA look, etc. 
Any chance Alex has a probe already calibrated to go that low or did we order any of those turbidity and conductivity probes from in-situ (the ones we looked at ISO the ploidy project?)

pmcelhany, 2022-06-10:
@KROVINSKI @dmperez89 Unfortunately, Alex does not have a good conductivity probe. Do we have enough conductivity standard for seawater. Might as well stock up if we are going to the hassle of ordering standards.

pmcelhany, 2022-06-10:
@KROVINSKI @dmperez89 alex can run mass spec water samples. We should collect them in the acid boiled tubes so they are trace metal clean (Danielle knows how to do this enjoyable process). We want samples of our homebrew water after the pressure tanks but before the DI filters, our homebrew after the DI filers, tap water from the pilot plant lab, DI water from tap in the main building and milli-Q water from the genetics lab. As soon as we have all five samples, we can run them over to Alex's lab next time someone goes to UW.

dmperez89, 2022-06-16:
If we're cleaning more tubes, we need more trace metal grade hydrochloric acid. Maybe Kathleen can easily find the same kind we used before?

pmcelhany, 2022-06-17:
@dmperez89 Yes, please ask Kathleen to order more acid.

pmcelhany, 2022-06-24:
tasks moved to multiple new issues

closed 2022-06-24 by pmcelhany