NWFSC-OA-lab/asw #27: Install salinity control system

pmcelhany, 2022-08-15:
- [ ] Install the high salinity pump 
- [ ] Mount conductivity sensor for head tank
- [ ] Install solenoids
- [ ] Install computer for feedback control
- [ ] test system

pmcelhany, 2022-10-07:
@michaelmaher114 @dmperez89 How close are you to getting the head tank conductivity meter operational? It would be great to get that online even if the feedback system is not fully operational so we we can test that conductivity sensor along with the orion probe as we figure out standard seawater.

michaelmaher114, 2022-10-07:
Hey Paul -

I am pretty close!  There have been a few back and forths with NI and ATI
but this has been my focus.  I think I should just about have it able to
record data by the end of today or early next week.

Mike

pmcelhany, 2022-10-07:
Great!

pmcelhany, 2022-11-14:
comm card for salinity probe is ordered