NWFSC-OA-lab/asw #29: Make seawater app

pmcelhany, 2022-12-13:
@dmperez89 @KROVINSKI 
I made a shiny app for making seawater cals in the [this repo](https://github.com/NWFSC-OA-lab/asw). The app is called make_seawater_app.R. It calcs how much salt and Di water to add to the HST that already has some saltwater in it and it calcs how much HST  and DI water to add to the head tank that already has some saltwater in it.

pmcelhany, 2023-05-08:
@dmperez89 @KROVINSKI 
Did you use this seawater calcs app?

dmperez89, 2023-05-08:
@pmcelhany @KROVINSKI I believe we used it for making larger batches of water, but if we only needed to adjust the salinity minimally (usually due to evaporation) we would simply add ~1 L DI water.

pmcelhany, 2023-05-08:
Sounds good.

closed 2023-05-08 by pmcelhany