NWFSC-OA-lab/asw #31: New salinity probes for ASW system

pmcelhany, 2023-01-19:
- [ ] Research another option for the head tanks and HST

pmcelhany, 2023-01-19:
@michaelmaher114
Since the vendor for the ATI conductivity sensor is so flakey, we should look at other options. What about just getting a couple of more Orion star A322 probes? It has a usb/rs232 interface - maybe we can live stream data and connect it to labview for data recording and system control. Can you take a look if that is possible? It would be nice to just have one system to master in the lab and they are versatile because we can use them as portable meters if needed

michaelmaher114, 2023-01-19:
I was just emailing with ATI and I asked if we could at least get the
communication card right away.  Hopefully that is possible and perhaps
I can cancel the rest of the order.

I will look at connecting the Orion star probe to labview.

pmcelhany, 2023-01-19:
@michaelmaher114 
If ATI is not going to come through, we should ask for a refund and put the money toward a different system.

pmcelhany, 2023-08-21:
Part of aws review issue

closed 2023-08-21 by pmcelhany